winder/Universal-G-Code-Sender #442: GRBL Not finish booting

jpilldev, 2016-07-21:
Hi.

I have an arduino Mega with GRBL hex file loaded, i open the connection but never gives promt that shows the GRBL version.

After some minutes i cannot see any advance

I try to send a gcode file but i got the error message.

Any suggestion to advance in this?

Thanks
[image]
[image]

winder, 2016-07-21:
I believe some Atmega2560 builds floating around have a boot message which can't be parsed by UGS, which leads to this error.

Try the version here: https://github.com/gnea/grbl-Mega
Or run the nightly build using the "Smoothie" firmware profile (identical in almost every way to the GRBL profile).

Let me know if that doesn't resolve the problem.

closed 2016-07-21 by winder

jpilldev, 2016-07-21:
Hi Winder.

I upload that example and run with Universal Gcode Sender V 1.0.9 (i didn't want to wait unit V2.0 build)

This solves the problem, now i can see the log like this:

`**** Connected to /dev/ttyUSB0 @ 115200 baud ****

Grbl 1.0d ['$' for help]
Skipping command #0
Skipping command #1
Skipping command #2
Skipping command #3
Skipping command #5
Skipping command #7
Skipping command #8
Skipping command #17
Skipping command #18
Skipping command #19
Skipping command #33
Skipping command #34
Skipping command #35
Skipping command #47
Skipping command #48
Skipping command #49
Skipping command #71
Skipping command #72
Skipping command #73
Skipping command #95
Skipping command #96
Skipping command #97
Skipping command #115
Skipping command #116
Skipping command #120
Skipping command #121
Skipping command #122
>>> M3
>>> G21
>>> G00Z5
>>> G00X32.6193Y12.0078
>>> G01Z-0.125F100.0
>>> G01X31.1089Y7.9122Z-0.125F400
>>> G01X34.1351Y7.9122Z-0.125`

Is this a good output?

Thanks

winder, 2016-07-21:
Looks good!