winder/Universal-G-Code-Sender #443: GRBL not responding UGS

AqsaAli, 2016-07-25:
hey!!
I connect my CNC xPRO V2 Controller Stepper Driver with system and start a software. but it does not select the USB port. and if it selects the port then GRBL in the shield does not respond.
When I look at the command table , the sent column is checked in front of all commands. but the done and responce column remain unchecked showing no command is done. and there is not even a jerk in the stepper motors.
How can I resolve this problem

winder, 2016-07-31:
You're probably using the wrong baud rate, try 115200.

closed 2016-07-31 by winder

AqsaAli, 2016-08-02:
I tried this also.. but there is no responce

winder, 2016-08-02:
AqsaAli, maybe contact the xPro team and see if the board isn't flashed or setup propertly. It sounds like you're using UGS correctly.