winder/Universal-G-Code-Sender #444: Limit switch issue

jahnj0584, 2016-07-27:
Ive switched to a new machine with the same gshield and limit switches. I wired it up like the old machine, with them being NC (ground on pin 2, other on 1 (C and NC)) and homing now doesnt work. 

With the multimeter, Z and X switches read ~1 and then drop to nothing when I click the switch. The Y goes from 1 and only drops to -2 when I click the switch (faulty switch?)

When I home, the Z will just rise and crash into the switch and I get an alarm. 

Any ideas? Is it just the X causing mayhem? Is there any way to see in UGS if a switch is actuated?

faultylee, 2016-07-28:
Feels like wiring problem if you didn't change the settings. It should either be 4~5V dropping to 0V or the opposite. 1~2V is weird. You might also want ask here https://github.com/grbl/grbl

winder, 2016-07-31:
I think feedback on which switch is activated is coming in grbl 1.0... for now I'm not sure what UGS can do to help troubleshoot this sort of problem.

closed 2016-07-31 by winder

jahnj0584, 2016-07-31:
I only have an analog meter now and I think I was using an Ohm setting....I still gotta watch a few YT vids on how to use these properly and have a digital one in the mail. 

I was just wondering if there was a command I could send that would be like limit switch (pin X) = activated/no signal

faultylee, 2016-08-03:
I guess you can always upload a simple Arduino sketch to read the limit GPIO pins. Make sure to configure the input as per your Grbl config (Pull up or external pull down).

jahnj0584, 2016-08-03:
I have 0 knowledge with arduino, besides sending the light blink sketch to
make sure it works when you get it. Is this something someone here could
write up?

faultylee, 2016-08-04:
If you can wait, I don't mind to hack one up over this weekend. For the meantime, you should try ask in other forums, such as http://www.shapeoko.com/forum/viewforum.php?f=3 

Btw, you might need some basic electronic skill to fix the actual problem. If you can find a local buddy to help out, that would save you a lot of time.

jahnj0584, 2016-08-04:
I have replacement switches here and a multimeter in the mail. If you
could, that'd be great! I won't be able to mess with them until Monday
anyways.

For what I do run now, I just reconnect the machine at my work 0 and start
it from there.

faultylee, 2016-08-04:
Do you mind to share you setup in more detail? Btw, I've been using my Shapeoko2 for 2 years now without using limit switches, don't really find the need to. I did set it up once to learn the "real" CNC workflow, but had some problem with the wiring after a while, and stop using it since. Any particular reason why you need to use the limit switches?

jahnj0584, 2016-08-04:
Mostly for templates and jigs/repeat operations as I do small (2-10 piece)
production runs. There also the G28 at the start of my Post that causes
issues if you don't have homing as a possibility.

I'll be doing two sided milling soon and need them then for sure.

COEP-Anasane, 2021-11-13:
We want to add limit switch to a X-Y plotter. We are using Universal GCode Sender Version 2.08.

$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)

are set as shown.
Still it is not taking the limit switch input to stop the process of plotting.
Can someone help us?