winder/Universal-G-Code-Sender #445: empty firmware selection - aug 4th build

jahnj0584, 2016-08-05:
Thanks for fixing the connection issue yesterday, but down the firmware drop down menu is totally empty and connecting without is just says bad config for GRBL.

jahnj0584, 2016-08-10:
Is this a WIP (was never notified)?
Ill download it again and see if it works. Just a few more wrinkles to iron out!

winder, 2016-08-11:
It's still a WIP, but it should be usable now. I didn't want to close the ticket (again) without doing some more testing.

jahnj0584, 2016-08-11:
I don't have any cutting to do until Monday but i will try it today and let
you know. I don't know if it's a good or bad thing that I'm submitting 99%
of the tickets these days :P

faultylee, 2016-08-15:
In my day job, we have 2 full time testers to test our application. Now @winder got a free tester, guess he wouldn't mind at all :p

jahnj0584, 2016-08-15:
I think this issue is solved. The one i downloaded today had a firmware and
even a connect button! :P

closed 2016-08-18 by winder