winder/Universal-G-Code-Sender #447: Universal Gcode Visualizer Graphics

DOUG888, 2016-08-08:
I am running the latest version of GRBL and latest version of UGS.
Windows7
The problem is the graphics in the visualizer are very rough, picture attached.
Cannot see anywhere to change their settings.
Help please.
 
[image]

winder, 2016-08-08:
Does it look like that with the platform build as well? Are your graphic drivers up to date?

jahnj0584, 2016-08-08:
That's how it looked on all my computers too. I assume it was designed that
way? Use platform if you want a better visualizer

winder, 2016-08-08:
It's supposed to look like in the screenshots. For some reason the line width gets messed up like this sometimes, I've only seen it on a Windows machine.

DOUG888, 2016-08-10:
OK...got the platform version up with latest Java 1.0.0_102, have run the above program and the visualizer is displaying correctly.
Problem#1...the Visualizer tool does not move while the code is running.
Problem#2...the DRO does not update...all axis display 0,0,0
these may be related, is a GRBL setting incorrect???
I find I have to $X unlock to get movement, this is common for other senders as well eg : GRBL Controller
I am new to GRBL CNC, only 1 week, previously ran Mach3 then changed to Linux CNC, now running GRBL, which seems to give better results than the other systems. Am currently re-wiring my machine and am using a CNC Shield V3 and some loose steppers etc to simulate.

DOUG888, 2016-08-10:
AHHHHHH!..please disregard previous post...I don't know what i did but the visualizer is now functioning and the DRO is updating ,readout seems to the correct values but is displaying in RED and occasionally BLACK, what is the significance of this please??
In the Windows Drop down, what is the Output supposed to display, Console is displaying a commented readout of the code, but there is nothing in the Output window.
In Console there is a comment "IOException while sending status command.: null" Is this an error or is all good?
A question I should have asked at the beginning :- Is the a tutorial on operation/properties or similar available?

jahnj0584, 2016-08-10:
Red just means that axis is in movement. You can turn it off in the
options.

DOUG888, 2016-08-10:
Yes..that makes perfect sense..thanks

winder, 2016-08-11:
If I were to update the classic gui's visualizer, it would probably be by porting the code from the platform. Closing the ticket because this bug probably wont be fixed.

closed 2016-08-11 by winder