winder/Universal-G-Code-Sender #448: UGS not loading file over 200kb

jahnj0584, 2016-08-09:
So, I have a file with 6 operations in it. The file itself is 214kb, and about 8300 lines I've code (Ive run longer 3D files, like 17k+ lines of code) but it wont load in the visualizer. If I load some smaller files before, this large file will load as a graphic for a flash second, then the lines remaining goes to -89 or something. I eventually had it run without crashing the program, and it did run but there was no timer and the whole progress window was error'ed.

I know I could split this up, but if UGS wants to be a strong program I dont think this should be an issue.

winder, 2016-08-09:
Looks like there has been a regression. I have a 26M file with almost 1 million lines of code which works fine in 1.0.9. That file opens in the classic gui but the visualizer doesn't work, and breaks the platform completely.

jahnj0584, 2016-08-10:
I assume thats just for testing purposes and not to actually cut something? 

Any leads on what may cause it? Should I stick to the classic GUI or start splitting up my codes?

winder, 2016-08-10:
That large file is a real one that someone who does laser engraving sent me (those projects have a ton of gcode). In the past I've used a test file that was nearly 1Gig though.

The build script I'm using for the platform is using mostly default settings, one of those settings sets the max amount of ram at 64MB - this is almost certainly the difference between classic/platform. I still need to do some debugging.

DOUG888, 2016-08-12:
The file attached is a bitmap of a grey scale picture , from what i can see there is nothing peculiar about the code, just the size. 2Mb
[maralyn.zip](https://github.com/winder/Universal-G-Code-Sender/files/414889/maralyn.zip)

The file runs under Classic (not well) but not under Platform. Locks up..actually its still runs but extremely SLOWLY. appears to lockup.
There maybe a G code incompatibility with the setup parameters

Doug

winder, 2016-08-12:
I increased the minimum memory required to 256mb and removed the max (was 64mb!).

My 26mb test file and @DOUG888's file now load.

Thanks for the bug report.

closed 2016-08-12 by winder