winder/Universal-G-Code-Sender #449: Sharable Macro Groups

winder, 2016-08-11:
UGS macros are an easy way to add new features and make personal workflows. Some of these workflows would be useful to other people if there was an easy way to group and share them.

Would it be possible to have a central repository of macro's and enhance the panel to switch between groups and browse available macros?

Here's a quick mockup of what this might look like:
[image]

jahnj0584, 2016-08-11:
I'd enjoy this. Right now i only keep the three probe macros saved because
there isn't much room for more. A bit down the road though once the current
major things get solved?

winder, 2016-08-11:
This is just an idea I've been thinking about. I have no idea if or when I'll work on it.