winder/Universal-G-Code-Sender #452: Output Window

DOUG888, 2016-08-13:
UGS Platform latest overnight
Bottom of the screen is the Console window, all good, next to the is the Output window, this is empty, is this for future development or a bug or have i stuffed up a setting.
If so what should it display??
Thanks
Doug
Apologies if this has been asked before.

winder, 2016-08-13:
The output window is built into the framework I'm using. I was going to try replacing the console window with that built in output window, but thats pretty low priority for me. For now you can close the output window, it isn't used currently.

closed 2016-08-13 by winder

DOUG888, 2016-08-13:
Thanks for clearing that up

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus