winder/Universal-G-Code-Sender #453: Current feed override setting

jahnj0584, 2016-08-14:
As it stands, using the feed over ride in the option is a crap shoot. Sometimes it works, sometimes it doesnt and you never know when it will be activated. It'd be great if there was something as easy as "sending at: XXX%" underthe file browser or even in the send progress box.

Cheers

winder, 2016-08-15:
With the new Gcode Processor's there's a disconnect between some options in the "Sender Settings" and "Gcode Processor Settings". Currently some options must be ENABLED in the "Gcode Processor Settings" but configured in the "Sender Settings".

I think I can improve those settings.

winder, 2016-08-15:
Having a way to see exactly how the gcode is being modified regardless of how it is configured is a great idea.

winder, 2016-08-25:
I reworked the settings page. All the settings which effect the gcode are now in the `Gcode Processor Configuration` page. After changing a setting it should reload the gcode file so that any settings will be taken into account right away.

Let me know if the override settings still feel like a crap shoot.

closed 2016-08-25 by winder

jahnj0584, 2016-08-25:
I'll test it tomorrow!