winder/Universal-G-Code-Sender #455: Cant g91 after probe

jahnj0584, 2016-08-14:
I try to macro
1)G38.2 Z-10 F50
2)G91 Z1
3)G38.2 Z-2 F15
4)G92 Z19,34 
5)G91 Z-10 F3000
(probe down, retract 1mm, probe slower, set Z height, retract 10mm so i can take the probe away)

But no matter what, the G91 causes a PROBE ALARM and a very strong lock that needs a soft reset before the $X. I CAN jog up with the machine control but its not ideal as theres usually a larger jog height in there from moving to the probe.

DOUG888, 2016-08-15:
Ran your code with, as expected,the same results, used the modified version below.
G38.2 Z-45 F50;G01 Z2;G38.2 Z-4  F15;G92 Z9.8;G01 Z15 F50;
Seems to work.

winder, 2016-08-15:
Like Doug pointed out, it's probably best to specify the movement command you want to use for commands 2, 4 and 5.

closed 2016-08-15 by winder

jahnj0584, 2016-08-15:
but doesnt using a g1 just mask the issue with the g91? I also have the issue with UGS swapping between g90 and g91 from time to time.

jahnj0584, 2016-08-15:
I will try your macro today and see how it works out.

jahnj0584, 2016-08-15:
Nope. Doesnt work. Z goes down slowly, then at max speed tries to jog down further into the probe.

DOUG888, 2016-08-15:
Jahnj0584 .. I think the reason that the probe rapids down again into the job is that you have Z15 after Z19.34, the last Z is smaller than your G92 command so it must go further down... I notice you have added a G91 command at the front of the macro code, I added this and had unusual results, but without it the sequence works. Have another go and see what happens.
Yes, if there is a glitch with the G90 G91 commands that should be looked at.

jahnj0584, 2016-08-15:
See, thats why I needed to add the G91. I want it to probe, then go 15mm away from the probe so I can pull it away, and not try to go 4.35mm into it.

DOUG888, 2016-08-16:
had a look around the web and found this, have adapted the plate thickness to yours...also you had a -Z as the last entry, I replaced it with a positive this will pull the probe further away, both seem to work.
G91;G38.2 Z-10 F25;G90;G10 L20 P1 Z19.34;G91 G0 Z10;G90
OR using G92
G91;G38.2 Z-10 F25;G90;G92 Z19.34;G91 G0 Z10;G90
Let us know how it went.

jahnj0584, 2016-08-16:
I despise G90 mode, so I'll have to pass on those ones. I believe G10 L20
P1 just does the same as G92 anyways right? The macro edit I posted above
is working great for me now.

DOUG888, 2016-08-16:
Glad to hear you got it working and Yes G10 is the new G92,both are still active at this time.