winder/Universal-G-Code-Sender #456: Scaling issue and z-axis motion

irusujith, 2016-08-16:
While using this software, when i give 1 mm as input it is scaling about 6 mm in the output, both in X-axis and in Y-axis (Hardware is X-carve).
The Z-axis is going in the opposite direction for which i desire. And even during the operation Z-motor is not responding, in the sense it should penetrate down during the operation but it is staying only on one plane. It is neither moving up nor down.

jahnj0584, 2016-08-16:
You have to calibrate the machine using 110/111/112. Videos are on YouTube.

Either set a mask for your axis or swap the wire pairs for your Z. Sounds
like you also need to adjust your trim pots

<https://github.com/winder/Universal-G-Code-Sender/issues/456>, or mute the
thread

winder, 2016-08-17:
It sounds like either: your machine isn't wired correctly, your belts aren't tensioned enough, you have a loose pulley, your power supply isn't strong enough for your motors, or some other misconfiguration. Unfortunately there's a lot that can go wrong when setting up these machines. 

I'd suggest you create a question with your community of choice, or contact the inventables support. They should be able to help you better than I can.

Good luck!

closed 2016-08-17 by winder