winder/Universal-G-Code-Sender #457: I and J code errors

DOUG888, 2016-08-17:
I am receiving an error that involves the I and J commands

When i run the program it accepts line N09 but fails on line N14 and N19 etc.etc.

the error reads:-

An error was detected while sending 'N14 X3.856 Y4.191 I-.144 J0. F250.': error: Invalid gcode ID:36. Streaming has been paused.
Would you like to skip commands like this in the future?

I can see the obvious differences between them with the G17 G03 missing. is this the problem, i thought that those commands translated thru??

( FOR USE WITH GRBL BASED LASER MACHINES )
( PROGRAM NUMBER: 100)
( PROGRAM NAME: AZTEC CALENDER2.NC)
( POST:  LASER  )
( DATE: WED. 08/17/2016 , TIME: 11:52AM)
(MTOOL T1 D.2 C0. A0. H0.)
(JOB 1  CONTOUR)

N01 G94 G90 G80 G17
N02 G00 G90 G54 X4.173 Y3.868

N03 G21 (Metric Units)
N04 M9 (Red Dot Laser OFF)
N05 M5 (Laser Off)

N06 ( MSG,SHAPE 1: )
N07 S1000
N08 M3 (Laser On)
N09 G17 G03 X4.173 Y3.868 I-.144 J0. F250.    <<<<<<<<<<<<<<<<<<<<<<<<<<<<

N10 M5 (Laser Off)
N11 G00 X3.856 Y4.191

N12 S1000
N13 M3 (Laser On)
N14 X3.856 Y4.191 I-.144 J0. F250.            <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

N15 M5 (Laser Off)
N16 G00 X3.553 Y4.528

N17 S1000
N18 M3 (Laser On)
N19 X3.553 Y4.528 I-.144 J0. F250.

DOUG888, 2016-08-17:
Leave this closed for now..it could be my POST processor that is the problem

closed 2016-08-17 by DOUG888

winder, 2016-08-17:
N11 changes the implicit movement word to G0, so when you reach N14 I and J
are not expected.

DOUG888, 2016-08-17:
Hi Will

Yes, I did some more research on the subject (that’s why I closed the original post) and found that out. Have posted a new report that give better detail and confirms what you have just told me , I’m hoping there is a fix for this as different Cam software will produce this type of script. And errors

Thanks for the prompt reply.

Doug

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus