winder/Universal-G-Code-Sender #458: Reading Arc move Errors

DOUG888, 2016-08-17:
It appears that UGS requires all incidence of I and J moves to be preceded by either a G01/G02 or G03.
With programs like BobCadCam etc. the initial noting of the G01 (2 or 3) is not repeated every line but is only added after the incidence of a different G02/G03, this seems to cause an error and stops UGS.

see an example below
N21 starts with a G02 which is reflected thru lines N22-N26 where it changes to N27 G01 then N28 G02 again. Where ever there is a missing G01/G02/G03 on an I J line there is an error and UGS stops.

I have run script where the G code is in front of every I J line and it work perfectly.

Can this be fixed to support different cam software.
Thanks

`N20 G01 X200.587 Y71.96
N21 G02 X200.392 Y73.038 I4.107 J1.3
N22 X200.912 Y75.04 I3.432 J.177
N23 X201.729 Y75.979 I3.093 J-1.866
N24 X201.98 Y76.163 I2.027 J-2.499
N25 X203.664 Y76.68 I1.718 J-2.594
N26 X204.824 Y76.536 I.047 J-4.368
N27 G01 X204.828 Y76.535
N28 G02 X206.794 Y74.417 I-.841 J-2.753
N29 G01 X206.795 Y74.413
N30 X206.838 Y74.175
N31 G02 X206.823 Y73.442 I-1.819 J-.328
N32 X206.485 Y72.908 I-.918 J.206`

winder, 2016-08-17:
This is part of the linux cnc gcode specification that GRBL implements:
http://linuxcnc.org/docs/html/gcode/overview.html#_commands_and_machine_modes

If you have specific questions about why maybe try asking on the grbl github page.

closed 2016-08-17 by winder

DOUG888, 2016-08-17:
Hi Will..Thanks..checked the link out.

also

I ran the attached code on UGS and got the expected error, however I ran it on GRBL Controller 3.6.1 and also on JCNC without the error occurring.

This would indicate that the problem is in the sender rather than GRBL. My experience with Linux source code programming is minimal but maybe

inserting a flag to capture the last G code used in IJ instances would solve the problem.

 

??are you running a Linux machine to test or a Windows, I am running Windows7. Would it make a difference?

 

I am sending this direct to you as I don’t want to create discord on the forum and figure we can sort it

As always 

Thanks

Doug

 

http://linuxcnc.org/docs/html/gcode/overview.html#_commands_and_machine_modes

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

winder, 2016-08-18:
> and_machine_modes
> u2lnLwE2hks5qgxaQgaJpZM4JmEgz> .Image removed by sender.

On the gcode controller options page do you have any commands filtered out?
I'm wondering if you clicked "yes" to one of those "skip this command"
errors:

DOUG888, 2016-08-18:
UGS Platform

Controller Options….no , nothing resembling G02/G03 only a G43H1Z2.54 and an M18

I have now cleared all that out.

However I just noticed that Arc expander is un ticked as is Feed speed scaler, these are as per default settings.

 

Tools/Options/General   Use System Proxy

                                UGS/Sender Options   Enable single step mode is un ticked.        

All other settings are default.

 

You may not have seen it yet but I have a fix for this. Posted just 5mins ago…see SOLUTION for I and J error posting                        

 

https://www.avast.com/antivirus

DOUG888, 2016-08-18:
Just ran one of the offending programs and noticed that the fault happened, as you said, after a G00  line, which does cancel the G02/G03 link. 

This would be a POST weakness. NOT a UGS fault……..  :p   (sorry about that)

So the solution Post I’ve put up is even more relevant than before as it is focused on the POST mod for my setup

 

https://www.avast.com/antivirus

DOUG888, 2016-08-18:
> Yes I saw your post. This is still puzzling to me, since I ran the same
> gcode you posted and didn't have a problem. Feed scaling and arc expanding
> are disabled by default so it looks like your filters are fine.
>
> I finally remembered while typing out this response... I found a bug in the
> gcode processor on the 12th that causes this. I'm really hoping your copy
> of UGS is older than that!
>
> If you speak regex:
> - Pattern GROUP_PATTERN =
> Pattern.compile("([GgMmSsTt].*?)(?=[GgMmSsTt]|$)");
> + Pattern GROUP_PATTERN = Pattern.compile("(.*?)
> ([GgMmSsTt].*?)(?=[GgMmSsTt]|$)");
>
> The "command splitter" attempts to split up the different commands so that
> I can report errors more accurately (i.e. grbl would report an error for
> line "G21 G90 G61" but you wouldn't know which command was the problem).
> The green part was missing which caused all implicit commands to be
> removed!!!

You must be State side…Your hours are reversed to mine…

 

No, I don’t speak Regex, only Batchee  ;p

But I’ll take your word for it.

 

I have made my posting more specific regarding the reason for the errors I was getting. It is specifically about the Post Processor.

 

This is the only version info I could find

Product Version: NetBeans Platform 8.1 (Build 201510222201) 

Java: 1.8.0_102; Java HotSpot(TM) Client VM 25.102-b14 

Runtime: Java(TM) SE Runtime Environment 1.8.0_102-b14 

System: Windows 7 version 6.1 running on x86; Cp1252; en_AU (serverlaunch) 

User directory: C:\Users\Goblin\AppData\Roaming\.ugsplatform\dev 

Cache directory: C:\Users\Goblin\AppData\Roaming\.ugsplatform\dev\var\cache 

https://www.avast.com/antivirus

winder, 2016-08-18:
Does your original file work with the current nightly build of UGS? Does it work with other gcode senders?

DOUG888, 2016-08-18:
No it still has problems with the latest Nightly version, Yes it does run under JCNC and GRBL Controller.
It still appears to happen after a G00 command when the G02/G03 Arc is before the G00 and the arc code continues after. Code attached shows this...first error set is at N232-235..next set is at N289,291,292,294 then at N299-302 etc

[Joe2006-Nest-500thick.zip](https://github.com/winder/Universal-G-Code-Sender/files/425988/Joe2006-Nest-500thick.zip)

winder, 2016-08-18:
Ok thanks for bearing with me. Maybe those other senders continue to send
after an error.