winder/Universal-G-Code-Sender #459: Error on the standard command

80UqyQCxh, 2016-08-17:
Uploading a file to the work of 2.3mb.
Over 3 hours 3 errors appeared quite standard commands. Here is an example: ". An error was detected while sending G01 X8.5625 Y54.9375 error bad format Streaming has been paused" The same was the case with M3 S80 command. 
In version 1.0.9 UGS had no such problems.

Used nightly build 614, Linux, GRBL1.0c. 
sry for photos
[image]
[image]

80UqyQCxh, 2016-08-17:
`G00 X22.375 Y54.9375
M3 S80
G01 X21.5625 Y54.9375
M5
G00 X21.125 Y54.9375
M3 S80
G01 X20.875 Y54.9375
M5
G00 X19.5 Y54.9375
M3 S80
G01 X18.9375 Y54.9375
M5
G00 X18.8125 Y54.9375
M3 S80
G01 X18.5 Y54.9375
M5
G00 X18.25 Y54.9375
M3 S80
G01 X17.8125 Y54.9375
M5
G00 X17.625 Y54.9375
M3 S80
G01 X17.3125 Y54.9375
M5
G00 X15.9375 Y54.9375
M3 S80
G01 X15.6875 Y54.9375
M5
G00 X15.5 Y54.9375
M3 S80
G01 X15.3125 Y54.9375
M5`

how look this gcode

winder, 2016-08-17:
The gcode you attached gives me the error "Undefined feed rate", it works fine if I add F100 to the first line.

This may be an issue with the GRBL 1.0 branch, could you ask this question there?

80UqyQCxh, 2016-08-17:
`G21; Set units to millimeters
G90; Use absolute coordinates
G92; Coordinate Offset
G00 X40.6875 Y1.0
M3 S80
G01 X41.875 Y1.0 F900
M5
G00 X42.375 Y1.0625
M3 S80
G01 X40.25 Y1.0625
M5`
6 stroke in my .nc

winder, 2016-08-17:
Also, if you are willing to contribute translations, it is much easier to do now than it was in the past:
https://poeditor.com/join/project/2J2hB5I41Z

winder, 2016-08-17:
I think your gcode looks good. It ran for me with grbl 0.9i

duisenberg, 2016-08-18:
I ran into a similar issue lately. It is not easy to catch, in my case it was the status request that happend to come in together with the gcode. 

Would suggest to modify the ugs error tab from ignore yes/no to yes/no/resend since the error is catched by firmware (so it doesnt lead to unwanted behavior) and it would save lots of time by just resend the command (and continue the job) instead of cutting out the already processed commands from the file manually and then resend it - what is quite nervy - or even worse restart the whole job.

winder, 2016-08-18:
@duisenberg how were you able to figure out that a status request was colliding with the gcode?

Do you mean UGS was sending something like `G0 X10 ? Y10`? or does the response have an error in it like `ok<grbl_status_message_here>`?

80UqyQCxh, 2016-08-18:
Today launched the long work on the 1.0.9 version. Everything is OK. I like the 1.0.9 version that write any gcode in the file, it still sends and even if error, the process will not stop.

duisenberg, 2016-08-18:
My firmware is not grbl, currently doing my own ... it justs mimics grbl comm protocol. Among some other debug output I send each cmd the firmware receives back as debug (in serial com what you send is not always what you get). Currently Im running some tests to reproduce this issue, but not successful so far. The only thing I can see so far is that from time to time some of my debug output is not received properly by ugs:
ARC: doLine(ccw-angle:0.46 Deg:26.47 Px:41806 Py:68359 nX:41796 nY:68364)
.46 Px:41796 Py:68364 nX:41786 nY:68369)
ARC: doLine(ccw-angle:0.46 Deg:26.46 Px:41786 Py:68369 nX:41776 nY:68374)
ARC: doLine(ccw-angle:0.46 Deg:26.45 Px:41776 Py:68374 nX:41767 nY:68379)
ARC: doLine(ccw-angle:0.46 Deg:26.44 Px:41767 Py:68379 nX:41757 nY:68384)
68389)
ARC: doLine(ccw-angle:0.46 Deg:26.43 Px:41747 Py:68389 nX:41737 nY:68393)
68398)
ARC: doLine(ccw-angle:0.46 Deg:26.42 Px:41728 Py:68398 nX:41718 nY:68403)

This is with 10 status requests per second. Set to 2 requests per sec the debug output 
comes in all fine.

duisenberg, 2016-08-19:
Have found a way to produce this issue. Setting poll rate to 100ms and then simply pressing the Y+
button: 

>>> G21 G91 G0  Y1
DEBUG: CMD=G2(2).
error:[unspecified error.]

In this case the received command was incomplete

>>> G21 G91 G0 Y1          
DEBUG: CMD=G21 G91 G0 Y1(13).
G0 goto X:0 Y:5095
ok

Sending the command from the command line results in normal  action with ok response
Setting the status request to 1/sec this issue is gone.

breiler, 2026-10-31:
I am closing this since it stale (10 years) and a lot has happened to the streaming logic.

closed 2026-10-31 by breiler