winder/Universal-G-Code-Sender #460: Whitespace & Arc expander

duisenberg, 2016-08-18:
Having whitespace remover unchecked will leave the whitespaces in place, as expected. Unless, if the Arc expander is checked, the gcodes from expander are coming without whitespace. (Classic interface, nightly build Aug. 14, Windows 7) 

Also the Arc expander cannot be set to steps of less than 0.2mm leading to quite inappropriate arcs of 1mm or under. The minimal step should be 0.02mm here. 

I'm currently doing a project for a mill firmware for small unipolars, ugs is the sender of choice here. Great work btw!

winder, 2016-08-18:
You should be able to type any number into the "Small arc segment length" setting, are you seeing 0.2mm arcs being generated when setting this to 0.02mm?

The arc expander happens to generate brand new code without spaces in it, so there are no spaces to be left in place.

If you have any specific requirements for your firmware send me a message - The UGS controller classes can be extended.

winder, 2016-08-18:
Nevermind, it seems that field switches itself back to 1mm after modifying it. That should get fixed.

jahnj0584, 2016-08-22:
what exactly is white space and what is the remover doing?

winder, 2016-08-22:
Whitespace is things like spaces and tabs. They aren't needed and they take up space that could be used for other commands in grbls buffer.

winder, 2016-08-25:
This should be resolved now. The configuration is now all done in the `Gcode Processor Settings` and can be set to any precision.

Let me now if you see any problems with the new settings style.

closed 2016-08-25 by winder