winder/Universal-G-Code-Sender #462: Limit Switches/Homing---Definitive answer please

DOUG888, 2016-08-20:
I have been studying the forums for a limit switch / homing setup over the last week or so for the latest and hopeful all future versions of GRBL. There is a LOT of opinions out there...Please , what is the recommended setup:-

NO switches in Parallel
NC switches in parallel
Pullup resistors, what value
Pulldown resistors, what value.
A posted wiring diagram and set of instructions would be great posted as a Sticky Note at the top of the Forum as this appears to be a VERY COMMON question. (pls note I am not shouting  :p )

What GRBL settings:
I assume $22=1
$20=??
$21=??

I would like to incorporate Hard Limits as I have used these successfully on another setup.
I run a generic 3 Axis Mill that will also run as a 2 Axis Laser.
I have 6 limit switches, 2 per axis, currently wired in parallel, NO connected to Pin and Ground. NO resistors or capacitors. Shielded cable.
3 Axis Homing will work for both these setups, I see NO conflict at that level.

This is a diagram that i came across, is this correct????
[image]

jahnj0584, 2016-08-21:
I do them as NC and no resistor. Just a common ground and then 9/10/11 (or
10/11/12?) whatever the pinout is for the arduino.

faultylee, 2016-08-22:
I used NC in series for safety reason, i.e. it triggered if any of the wire within the loop breaks. I also used external pull down to avoid straining the internal resistors for long hours of operation. FYI, I seldom shutdown. Had to add caps on the limit pin to reduce false trigger from the motor noise.

DOUG888, 2016-08-22:
Ok...Solved...NO switches in parallel, 2 per axis with .1uf cap between pin and ground, no resistors, no 5vdc to anywhere.
I realize that wiring series has benefits but i will be there when HOMING is done.
I also enabled a line in config.h that sets the DRO output, it now shows 0,0,0 when HOME is complete instead of _5,-5,-5

#define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.

closed 2016-08-22 by winder