winder/Universal-G-Code-Sender #465: SUGGESTION Return to Zero Overnight #620

DOUG888, 2016-08-26:
Just ran the latest Overnight UGS Platform. #620
I was part way thru a test code and was satisfied all was good, pressed STOP (red square) and then pressed "Return to Zero" , what happened was that i was in a negative Z at the time and the machine (fortunately my mechanical simulator not the actual machine) ran the X and Y to zero then the Z, if i had been on my actual machine this would have been a disaster, obviously i should have looked and jogged the Z above the work piece before hitting the "Return to Zero" but this is something that could easily happen to anyone.
I don't know if this is new with this version or it's always been there.
Suggestion....When the "Return to Zero" is pressed, that it moves to a safe Z or at least to a Z clear of the work surface, this could be a setting in "Options" and could save the machine from idiot moves like mine.   :p

winder, 2016-08-26:
Unfortunately this feature was having some problems so I removed it last month. At some point maybe it will be re-added but for now it seems like there are too many edge cases to safely implement it.

Reference:
https://github.com/winder/Universal-G-Code-Sender/commit/36a28d93c499dfd407300e795769e803def55407#diff-1b227c000fe39010138854fc71fc575cL186

closed 2016-08-26 by winder

jahnj0584, 2016-08-26:
I think it'd work well if it just moved to machine Z0 first, then went to
the work X/Y and then Z.

winder, 2016-08-26:
I'm guessing the issue was related to the way GRBL sets the zero location after homing. It was causing a lot of problems though.