winder/Universal-G-Code-Sender #466: Single Step issue

duisenberg, 2016-08-27:
In single step mode, the sender should send a command and then wait for answer. If ok, the next command can be sent, if error the sender should pause the transfer and issue the "skip this command in future" dialog box. 

As it is, the sender sends the next command before pausing the transfer:

[image]

In the above case my firmware (emulating grbl) issues an error at M6 to force ugs to pause transmission, enter its toolchange mode where it moves the table into a position the tool can be changed and after change is done, the next M6 (manually entered or from file) would drive the table back into its last position.

Currently I work around this by sending M1 before an M6 what - since M1 produces an error too to force the transmission to pause - works. But it would be better, if UGS in single step mode acts as expected.

winder, 2016-08-27:
Nice catch, I'll try to fix this soon.

winder, 2016-09-02:
This should be resolved now, let me know if you see any other issues with single step mode.

closed 2016-09-02 by winder

duisenberg, 2016-09-04:
Looks good, currently its purring like a kitten.  If other issues come up I'll keep u posted.

commit 635dfedf (2016-09-02): Stop sending commands in the communicator if an error is detected. Closes #466