winder/Universal-G-Code-Sender #468: Translation problems

cmsteinBR, 2016-08-27:
Dear,

Some strings are not being translated (remain in English in the program GUI):

mainWindow.swing.activeStateLabel
gcodeTable.command
gcodeTable.originalCommand
gcodeTable.sent
gcodeTable.done
gcodeTable.response
mainWindow.swing.showCommandTableCheckBox 

They are missing strings to translate the captions of the buttons in tab Machine Control:
Reset X Axis
Reset Y Axis
Reset Z axis

Thanks

winder, 2016-08-27:
I fixed these:
mainWindow.swing.activeStateLabel
Reset X Axis
Reset Y Axis
Reset Z axis

The rest work for me.

closed 2016-08-27 by winder

cmsteinBR, 2016-08-27:
ok. Please, see the figures.

Thanks
[image]
[image]