winder/Universal-G-Code-Sender #469: Translation problems

cmsteinBR, 2016-08-28:
Please, see the figures.

Thanks
[image]
[image]

ml-hn, 2016-08-28:
I'm from Germany and I using Windows 10. The GUI shows no special characters (Umlauts). These are all displayed as squares.

Thank You

winder, 2016-08-29:
@cmsteinBR Thanks, updated these fields. There is a new one for `Pendant` that you can add on POEditor.

@ml-hn Sounds like the same issue as #315

cmsteinBR, 2016-08-29:
Other translation missing: caption of edit buttons in GCode Processor

breiler, 2019-03-02:
This has been resolved

closed 2019-03-02 by breiler

einencool, 2020-03-21:
Hello,
I would like to ask if there is any possibility that I have to install a extra software, so that the german translation will be displayed correctly? I have the Problem, that many "umlauts" are displayed as a black square, and I don't know, what I can do in this case?

[image]

I've read about the translation and took a look  at the "poeditor" Page, and there are all translations displayed "correctly" like "Fräser Durchmesser" for "Endmill Diameter"

[image]

Maybe I'm missing something :-)

Greets
Chris

commit f3c6fa81 (2016-08-29): Fixes some localization fields. Closes #469