winder/Universal-G-Code-Sender #470: Suggetion---Work Zero vs Machine Zero

DOUG888, 2016-08-28:
I am not a Machinist so there may be a valid reason for this to be the way it is....When I hit $H to re-home my machine the Work co-ordinates set to ZERO as you would expect but the Machine co-ordinates retain a value, sometimes plus, sometimes minus.
I would expect that a Homing operation would set everything back to square one and Zero all values.
So my suggestion is that it does.

jahnj0584, 2016-08-28:
Yea, something is wrong then. Machine 0 should always be 0 after a proper
home, and work 0 will retiain the relative value of wherever you set it
before

winder, 2016-08-28:
Maybe this will help:
https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to

duisenberg, 2016-08-29:
Yep, exactly. It is not a sender issue, the sender just ready what the firmware sends on ? command.  The cnc firmware in homing will run the axes into negative space until the end switch fires. This point is defined at a specific position in parameters. Then it runs the axes back a bit until the switch releases. Due to the switches construction the engage and disengage positions of a switch differ a bit.  Both of this points should be in negative space so they do not interfere in common operation of the mill.

DOUG888, 2016-08-30:
Thanks Guys..I should have mentioned that I have a moving Table machine in the X and a gantry in the Y...i had a Home position in the Front Right now it is in the Back Right.
Not only did the changes that were suggested give me better access to the table it also gave me the minus 2 values in the machine position box (2 being the home step off size). Now the screen matches the job and matches the tool direction, even if I do a re-zero work values during a setup, when I Home I get the -2 back in Machine , a win on all counts.

closed 2016-09-01 by winder