winder/Universal-G-Code-Sender #471: Resume from Pause - i did not work

cojarbi, 2016-08-30:
Hi guys, i have read already many issues here regading this topic but i want to ask for a recommendation on how to proceed.

Im doing a 35000 line halftone on my cnc and had to pause it overnight at about 20000. Now hitting resume or any other command just did nothing, those command accumulated but no moves. 

I recorder the last Ok and its location from the machine so what would you recommend so i can continue from this? I know i can just copy the gcode from there but how to i get to the exact location? Mind that the spindle is at the last Ok but i disconnected the UGS session as nothing was happening 

Appreciate the help

duisenberg, 2016-08-30:
Yes, thats a bit messy atm indeed. With absolute coordinates G90 used you could easily split the file and resposition manually with a G0, but if its G91 is you havent written down the last position ... well, the only way I'd see here is to run the first part of the file in $C mode to retrieve the coordinates. 

It might be done a bit easier if UGS would allow machine control commands or command line in pause mode but this currently does not work. No idea if this really can be done in a sane way in grbl command stack mode, in single line mode it might be a bit easier, but I'd guess there is a reason for leaving the buttons disabled in pause.

winder, 2016-08-31:
Most likely the serial connection dropped overnight. A recovery mode (#123) is something I'd like to add someday, but it's not a high priority right now. The main trick is that in addition to start with the correct command at the correct location is that you need to have the correct modal state ($G).

As for running commands while a pause (GRBL feed hold), this would be tricky since grbl is stopping part way through a command.

closed 2016-08-31 by winder

faultylee, 2016-09-01:
@winder What about an option to pause at the end of the command, or even crazier suggestion, allowing breakpoint