winder/Universal-G-Code-Sender #472: UGS Platform

Encarnacao, 2016-08-31:
First would like to thank you for the great work you got it done here to share with our comunity ;)

dont know if would be a good idea but maybe could be add a button for spindle on and spindle off and maybe to something to predefine S1000 for people like me that uses dremels (just for testing options cause usualy u set speed on software you do the g-code)etc...
also when i browse for add a file to do a job i couldnt find where to start sending file untill finaly i seen the play button on top, would be great to have below where u browse file also a send button.

just a idea...

thanks alot

jahnj0584, 2016-08-31:
You can move the buttons around. Ideally you want it far away from
everything so you don't accidently send a job while trying to disconnect or
home.

Surely you can make a macro for S1000 and S0

Encarnacao, 2016-08-31:
true i can make a macro, what if we could add text in front of macro ex:

0        turn spindle home
1         goes to work position
2        go start point

etc...

jahnj0584, 2016-08-31:
You can do that already when you go to edit the macro

Encarnacao, 2016-08-31:
i see that but only shows if you go with mouse on top, would be perfect if showed next to it

winder, 2016-08-31:
You might need to restart the program to see the macro number display the name (Issue #454).

I think in the future these sorts of features could be added with "macro groups": #449

jahnj0584, 2016-08-31:
You're changing the wrong field. The one on the left side is the button
name, the right one is the hover description

Encarnacao, 2016-09-01:
Got it done mate i was defenetly doing on wrong field 
is there any dark themes avalible?

winder, 2016-09-02:
I haven't tried them, but NetBeans (and swing) have custom look and feels you can install and use:
http://plugins.netbeans.org/plugin/62424/darcula-laf-for-netbeans

If you give it a try could you create a new ticket to show how it works?

closed 2016-09-02 by winder

Encarnacao, 2016-09-03:
will do it...