winder/Universal-G-Code-Sender #477: position system question

Encarnacao, 2016-09-01:
Why in UGSPlatform lets imagen i make a macro G1 X100 Y100 Z7 when gets to position then if i say Go G1 X0 Y0 Z0 it will not do nothing, and if i say again in macro  G1 X100 Y100 Z7 will end up in G1 X200 Y200 Z14, is this a bug or is just how it works?

jahnj0584, 2016-09-01:
I hope these were all an error. G1 is a rapid linear motion. You have to
set G90 for absolute dimensions so when you type x200 it goes to x200 from
no matter where it is, instead of 200 more on the x like when you are in G91

closed 2016-09-01 by winder

Encarnacao, 2016-09-01:
yes you guys are right the problem was the G1 comand, setting to G90 solved the problem, i can see i have edit this post 2 or 3 times and end up with like 3 or 4 posts about same, dont see where i can delete them you free to delete, 

thanks alot for the info more im using the UGSPlatform more im loving it hehe :)