winder/Universal-G-Code-Sender #478: Netbeans crashing

jahnj0584, 2016-09-01:
I run the platform64 as admin or from shortcut (like I have for months!) and after a second or two it crashes. UGS never launches. The furthest I had it go was loading the start.base module, and it couldnt find it. I thought maybe GDrive lost a file in sync but even from unzipping a local archive from last week the issue is there. Reinstalled java and restarted, issue is still here.

winder, 2016-09-02:
Could you try running from the command line to see if there are any error messages?

Does `platform.exe` also crash or just `platform64.exe`?

There is also a settings directory created by netbeans platform that you can try clearing out, I'll need to lookup where that is located.

jahnj0584, 2016-09-02:
What is the command to do that?

Both.exes fail. The non-plat I have from July 21 launches right away. I'll
try a fresh download today if the Internet cooperates in the shop.

winder, 2016-09-02:
If you put `cmd` into the windows run box it launches the terminal application. From there if you put the ugs platform directory on your desktop `cd Desktop/ugsplatform/bin` puts you in the same directory as the executable (you can verify with the `dir` command).

Now run `ugsplatform.exe -J-Dorg.level=FINEST -J-Dcom.level=FINEST -J-Djavax.level=FINEST -J-Djava.level=FINEST -J-Dorg.netbeans.core.startup.NbInstaller.level=FINE`

jahnj0584, 2016-09-02:
Didn't work. Using the zip I downloaded from yesterday. Both normal and 64
do the same thing. I think this is an issue with my computer and not the
program (since old versions don't work either) but not sure what could have
changed? Compatability mode didn't do anything.

jahnj0584, 2016-09-02:
it launches on my main computer, but the window is blank so I guess that's an issue for another ticket.

jahnj0584, 2016-09-08:
Main computer works with Java 8. Shop computer gets to the loading modules (about 70% done in net beans) and then just crashes.

yllar, 2016-09-13:
try deleting/removing %APPDATA%\.ugsplatform and see if it works. also make sure you don't have older version of java installed

jahnj0584, 2016-09-13:
java was uninstalled when the new one installed. I'll check the folder, but
i already deleted any folder called netbeans on my drive.

jahnj0584, 2016-09-13:
didnt help. Still crashes as it hits the loading module part.

jahnj0584, 2016-09-13:
[image]
Does this help at all?

winder, 2016-09-23:
@jahnj0584 were you able to delete the directories created by the netbeans framework?

On my Windows 10 VM they are located here:
User directory: C:\Users\mac-vm\AppData\Roaming\.ugsplatform\dev
Cache directory: C:\Users\mac-vm\AppData\Roaming\.ugsplatform\dev\var\cache

jahnj0584, 2016-09-23:
In my user folder i have .ugs .jscc and .oracle-jce-usage

winder, 2016-09-23:
@jahnj0584 In that directory there should also be an AppData folder, on my VM it is a hidden file and I had to enable show hidden files to see it.

jahnj0584, 2016-09-23:
I found it by searching my C drive and deleted the whole .ugsplatform
folder. It launches now but as anticipated my macros and color theme is
gone. Wooooo!

jahnj0584, 2016-09-23:
this can be closed.

closed 2016-09-23 by winder