winder/Universal-G-Code-Sender #479: UGS Platform DarkTheme

Encarnacao, 2016-09-03:
Here is how it looks the DarkTheme of UGS Platform

https://photos.google.com/share/AF1QipMxg1UenB7mvBp1aao4Wcih0tNoAOLz-fl7q87E-lzK-nyDirYVwwajfr5nUv9SXQ/photo/AF1QipMXK4Um6yYA2ldjg9AAsrezyicuvfKrfer9rRXA?key=UnVPdVpLVzQzOHQtYVF1cnA3R19KeEpVM0dUb0pB

jahnj0584, 2016-09-03:
I like it!

jahnj0584, 2016-09-03:
where and how do you install it?

Encarnacao, 2016-09-04:
you download the plug in on
http://plugins.netbeans.org/plugin/62424/darcula-laf-for-netbeans
You go on UGS Platform tools / plug-ins / downloaded / add plug ins and you can see where is directory to drop the file you downloaded , it should be on C: / Users / name of your comp / AppData/ Roaming/.ugsplatform/ dev and u drop there file
to see the AppData u must put on top to see hidden files
on the UGS platform installed plug ins you can set as active or inactive

winder, 2016-09-07:
Thanks for sharing, good to know this actually works! I'll make a note to update the website: https://github.com/winder/ugs_website

closed 2016-09-23 by winder