winder/Universal-G-Code-Sender #482: visualize idea?

Encarnacao, 2016-09-07:
In UGS platform would be nice if would have like maybe some buttons with pre define points of view like front, back, side and isometric ?
also when zoom in,  we cant move to sides to zoom in specific points, would it be possible to do it?

jahnj0584, 2016-09-07:
I believe I already submitted this to snap to the Z cube faces. There's no
reason to view it from the side?

Encarnacao, 2016-09-07:
was just a idea sense when u zoom most like you be off the tool path and cant see, if you could with mouse also move to the sides wouldnt be bad

winder, 2016-09-07:
I want to have some buttons like this for preset camera angles. Sometimes I view from the side to check that tabs are created properly.

jahnj0584, 2016-09-07:
While we're talking about it can we add multiple side dimensions? Work
height, and safe heights (basically anything above or below the work Z0 as
a seperate value).

winder, 2016-09-07:
I guess I could look at which dimensions only have G0 movements to determine what the safety height is. As it is though, the Z label can be really small compared to the X/Y labels so it might be hard to fit in extra information. I can take a look at that when normalizing the label to point forwards.

Encarnacao, 2016-09-07:
cool
one more question does UGS platform will have auto leveling plug in on future?

winder, 2016-09-07:
Yes an auto leveling plugin is in the future, a lot of the code cleanup work over the past year was designed to make features like that possible. Thats probably what I'll work on once the first stable platform build is released (and I get my limit/probe switches wired).

DOUG888, 2016-09-07:
I'll second the ZOOM in to a location that Encarnacao mentioned, maybe focused around the mouse pointer would work, a lot of graphic packages work that way.

winder, 2016-09-07:
You can pan with shift+click and drag

DOUG888, 2016-09-08:
Thanks

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

winder, 2016-09-08:
I made a quick demo of some of the recent visualizer improvements. I'm still working on the select-to-zoom one at the end, but it should be finished shortly.

[image]

Encarnacao, 2016-09-08:
looking good , i  have tried the  pan with shift+click and drag works good, would it be possible to hold right mouse click and move with mouse, would be more easy acess?

winder, 2016-09-08:
I think the middle-mouse button + drag also pans. I'm open to suggestions.
Definitely an easy to find help box is still needed to describe all these
shortcuts.

jahnj0584, 2016-09-08:
UGH IT LOOKS SO GOOD
Whens release? This weekend? I have another batch of jobs next week so if I could play with it over the weekend that'd be great.

jahnj0584, 2016-09-08:
also good call on the right click to jog, should prevent some accidents.

winder, 2016-09-08:
Everything but "select-to-zoom" should be in the nightly build already!

jahnj0584, 2016-09-08:
Ok I'll have to transfer the new zip over tomorrow then. Been having net
issues in the shop computer lately

DOUG888, 2016-09-08:
Nice work Will, everything you mentioned so far working well on my machine. 
Tried moving/rotating and panning while running code, all good. Windows7 with 9j.
Thanks for Your efforts

DOUG888, 2016-09-08:
One thing I just noticed, i forgot to set my port to com5 on startup, which i operate on, so it was set to com1 default, but ran anyway. My guess is that this is a saved setting and it just shows com1 as a display but operates on the setting that i previously made. Am i correct????

If this is a bug, please don't loose focus on your current mission, i can wear it for a while.    ;-)

winder, 2016-09-08:
Could you create a ticket for that? Classic ugs saved the port but I don't
remember if I added it to the platform. I've been using the testing
firmware option lately for development which doesn't use that field at all.

DOUG888, 2016-09-08:
Done

DOUG888, 2016-10-05:
Will

Sorry to bother with a small question that has probably been asked before
but I didn't want to raise an ISSUE for it...when GRBL 1.1 or whatever comes
out will it support M06 M6 tool change.

Thanks 

Doug

 

Ps: have now got a working POST for Diode Laser that will step Z as
required, all it needs is a tool change ability to vary the Laser intensity
to complete it.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2018-03-18 by winder