winder/Universal-G-Code-Sender #484: UGS Platform with wireless controler?

Encarnacao, 2016-09-10:
i have seen some UGS controlers with the possibility to set a wireless usb joystick game pad option
not sure but i think wouldnt be a bad option to add, on the future, however i have tryed before using a joystick and i notice something i didnt like, i if you keep pressing to go oneside, when you release button it will keep moving, i guess that would be cause the information keeps feding the buffer of the arduino, is there any solution to bypass this problem and when u release it will just stop the movement?

jahnj0584, 2016-09-10:
Set the controller to be a single command and not hold the key. I use
keysticks and its fine

Encarnacao, 2016-09-10:
what program you using for assign the keys to the joystick?

jahnj0584, 2016-09-10:
As I said, keysticks but I have switched to Xpadder.

https://youtu.be/Y-erKT4zRlY

Encarnacao, 2016-09-10:
so you got to do a press for eatch step you do, if you hold key then u got no control where will stop

jahnj0584, 2016-09-10:
That's how jogging works... Unless you have it to 2mm or less, holding it
down will not stop immediately after you release the button.

Encarnacao, 2016-09-12:
what if u used the joystick instead of the keys of it,  maybe hack the joystick with a ic  555 to slow down the frequency that sends the impulse  maybe with the pot to adjust untill you find the right frequency of impulses and when you release would just stop, no idea if would work

jahnj0584, 2016-09-12:
you program the joystick to act like a seperate keypress+hold, and at 1mm jog it's moving in "real time". That is a fact of physics and doesnt matter if youre spamming the mouse or a joystick.

closed 2016-09-23 by winder