winder/Universal-G-Code-Sender #485: Homing switches triggered on start up

Kakrial, 2016-09-17:
When initially running a file, the machine moves to machine coordinates (0,0,0), however, due to the slight slop of the limit switches, they become engaged at X=-0.381mm Y=-0.381mm z=-0.054mm, just before it comes to rest. This stops the run as hard limits are engaged and cannot continue.

My question is, is it possible to change the machine co-ordinates to something like (-5,-5,-5) or to only allow the machine to go to that position before a run?

Cheers.

jahnj0584, 2016-09-17:
You should fix your limits, increase your debounce limits, or remove the
code that goes to machine home first.

<https://github.com/winder/Universal-G-Code-Sender/issues/485>, or mute the
thread

winder, 2016-09-17:
I think you'll have better luck creating a grbl github issue for this question.

You can also read this: https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to

closed 2016-09-17 by winder