winder/Universal-G-Code-Sender #487: Visualizer causes exception on Raspberry Pi

garyrkey, 2016-09-22:
I have a Raspberry Pi as a controller for a DIY CNC and use UGCS as the G-Code sender. Things work great as far as sending the G-Code is concerned, and thank you very much for developing this software. 
I do have a problem with the visualizer part, though. When I click on the Visualizer button, nothing happens in the UGCS screen, but in the terminal window used for starting UGCS, and exception is shown as in:
[image]

Is there a solution that you know of? I've searched for similar problems, but all seem to be from the 2013 era. 

My Pi is a version 3, Raspbian-Jessie, updated and upgraded. 
I have tried multiple versions of UGCS, including a couple of the nightly builds, latest being Sept 22, 2016.

Thank you,
Gary

winder, 2016-09-23:
Hi Gary, unfortunately this is a known issue. Something about the way I'm using the java 3d library (jogamp) is incompatible with the RPi. Does the visualizer in the UGS Platform version work?

garyrkey, 2016-09-23:
If you are referring to the ugsplatform64.exe, it work great on my PC!

Here's hoping that the Pi related answer comes to you in a flash of
inspiration. :-)

Thanks again for providing a product that lets me use the Pi in my control
box. It's a good thing! I can easily live with visualizing elsewhere, and
cutting with the Pi.

Gary

garyrkey, 2017-03-28:
Would it help resolve this problem if I sent you a Raspberry Pi (with power supply and Raspbian/Pixel installed) as a testing platform?

winder, 2017-03-29:
@garyrkey yes, that's pretty much the only way I'll be able to address this. But I'm a little hesitant to accept a donation since I'm not sure when I'll have a chance to work on this.

closed 2017-03-29 by garyrkey