winder/Universal-G-Code-Sender #491: SUGGESTION---Text dump of settings and Macro

DOUG888, 2016-09-27:
When the GRBL edit function is added, a button to dump the GRBL and Macro's to a ".TXT" file would be great, why you ask, well if you have to re-install GRBL everything is lost from GRBL and having a text copy of the Macro's is essential if you have a system crash and have to re-install the UGS software....ask me how i know this....  :-{

Thanks for all your efforts.

jahnj0584, 2016-09-27:
Its too late now, but I have all my GRBL settings and the probe macro in my
sticky notes. "Stickies" is the program I use and it backs up regularily to
a text file.

DOUG888, 2016-09-28:
Thanks Julius

I’ll have a look at that, what is the file name that stores the Macro’s etc. I may have a general backup that saved them….i hope.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jahnj0584, 2016-09-28:
the file is whatever you save it as. It wont archive the macro's you have in UGS but they will be a copy/paste away!

DOUG888, 2016-09-28:
OK…question…windows7….what are the names of the GRBL settings files and UGS (settings and macro) files that I need to backup and where would they be located

thanks

 

https://www.avast.com/antivirus

jahnj0584, 2016-09-28:
It would be whatever comes up after you send $$,and macros are whatever you type in

winder, 2016-09-28:
@DOUG888 if you have a backup with your old filesystem, UGS had all your macros saved to a file in your `%UserProfile%` directory. Depending on the version you were running it is either in a file named `.UniversalGcodeSender.json` or a folder named `.ugs`.

DOUG888, 2016-09-28:
Thanks Will…I will look for those.

 

https://www.avast.com/antivirus

closed 2016-10-05 by DOUG888