winder/Universal-G-Code-Sender #493: Code interpretation Problem

DOUG888, 2016-10-03:
Guys
i am trying to convert a POST Processor for Bobcam V24, but i keep getting unusual graphic results in UGS Platform.
I am posting here on the off chance that its a graphic issue as I can not see any Gcode error.
I have included all files and 3 pictures to help explain the problem....
Files:- Gcode for parts and Post .
Pictures:-
BAD is what i see with this Gcode.
GOOD is without the Z components excluded (Z is required for multi pass at varying depths to maintain focus)
Also note the partial circle at the ORIGIN.....not in the code at all.
BobCam picture is what i expected to see, this shows a correct tool path.

NB: I have not attempted to run this code as yet.
[Files.zip](https://github.com/winder/Universal-G-Code-Sender/files/505784/Files.zip)

??????
Your thoughts.
Thanks 
Doug

jahnj0584, 2016-10-04:
you may wanna change your post. It does not do the 3/4 circle at the start for me. It DOES move while your Z is down, and then go up the 0.25 before doing its first pass. You also have it as a cycle so it is cutting this part out multiple times in one file.

DOUG888, 2016-10-04:
My ISSUE has been updated

DOUG888, 2016-10-04:
OK...May have found the problem.
Line N04 contains a G90 and G54, I removed these and the partial circle disappeared, I tried each one individually, but, it has to be both removed to fix the display. The program run OK as well.
I don't understand why this is so, maybe someone with more experience can answer that.
(N02 T1)
N03 M05
N04 G00 G90 G54 X11.995 Y9.938
N05 M08
N06 Z2.54
N07 Z0.
N08 G01 Z-.25
N09 G03 X11.995 Y9.938 I0. J-2.175 F500.

closed 2016-10-04 by DOUG888