winder/Universal-G-Code-Sender #494: G4 P0.01 Problem

DOUG888, 2016-10-05:
I have an issue i can't solve...2 Programs attached:
Prog1....Displays correctly for the first part then all I can see is the Rapids..this file contains a G4 P0.01
Prog2....displays correctly...this file does not contain the G4 code.
both files cut correctly.

I have tried various values for G4 P___ but no luck.
I have run other code with G4 P successfully, but they had only a few instances.

Can a few of You run both codes thru UGS Visualizer and come back with the results please
Thanks
Doug
[Files.zip](https://github.com/winder/Universal-G-Code-Sender/files/509976/Files.zip)

breiler, 2018-07-27:
I had problems opening the files due to the line 12 in both files. Had to change the line:
```gcode 
N01 G00 G17 G21 G40 G49 G80 G90 
```
to

```gcode
N01 G17 G21 G40 G49 G80 G90
```

I think the files looks the same in the visualizer:
Prog 1:
[image]

Prog 2:
[image]

breiler, 2019-02-24:
Closing due to no feedback from issuer.

closed 2019-02-24 by breiler