winder/Universal-G-Code-Sender #495: SUGGESTION--Using the Tool Change ability already in GRBL

DOUG888, 2016-10-05:
I asked GRBL about M6 and this is the reply:
"From Sonny Jeon.......First, you can do M6 tool changes already, if you have a GUI that supports it. Grbl provides the basic commands to do this, namely a tool length offset command to alter Z-changes in tools. All a GUI needs to do is intercept an M6 command and insert a (customizable) macro program to do the tool change. There are a few GUIs that do this. bCNC is one that I know does this."

This is a ray of sunshine I wasn't aware of,
 1/. can this already be accessed in UGS Platform, if so how???
2/.  If not, can this be incorporated, Not sure how the "Customizable Macro" would work but I was thinking a popup frame with slots that lists tool parameters against a classic T number as most CadCam software tool lists do, so that UGS tool list can be directly related to the CadCam tool list.

winder, 2016-10-06:
Actually, the new gcode processor modules could do this pretty easily.

Could you expand on your second idea? I can see it working in several ways:
1) When you load a program you get a bunch of popups asking for the tool offset for each M6, then insert the macro (i.e. raise tool to Z, move to X/Y, pause, etc).
2) The offsets could be added to the end of each tool change like `M6 (offset: 1.23)`, then the macro and tool offset could all be processed automatically.
3) The code could be automatically split into separate files and the existing "workflow" plugin could be used.

DOUG888, 2016-10-06:
in Your 1). the inputs would cover all the normal tool requirements and some personally customized slots...for example:- in Laser operation, I have a series of Tools (Lasers) at different feed rates, having the ability to alter the S___ value on the fly for all the "T1,T2,T3,etc" would be a benefit, at present my POST can only specifie a single S___ value. This is hard wired into the POST, but with tool change active, it would then be an Automatic process.
Your 2+3) take the process thru the remaining steps.

UGS would look for M6, that would be the trigger,up comes a box with the pre-defined vales for that tool in it, where an M6 for a specific tool , say "T1" has been done , this would be stored and reused whenever a "T1" occurs in the Gcode and so on for all tools called.

I am not a programmer so i hope that this makes sense, all i can do is feed you a concept.
Doug

aforww, 2017-07-07:
Was there any headway on this? I was experimenting with using the M00 command to automatically halt the machine when an operation requiring a tool change occurred. However, there's no way of zeroing out the Z height as all controls are locked out.

TheNetStriker, 2019-12-03:
What is the status of this feature request? Ist this possible with the software now?

BrownTrout-coder, 2020-03-11:
What's happeneing with the tool change??? has anyone made a Gcode parser, which converts the M6 to a pre-storred macro??

kjk25, 2022-04-18:
all what we need is a macro thats right, please implemt it.