winder/Universal-G-Code-Sender #497: Is UGS Platform 2.0 considered as reliable as 2.0 Classic ?

splitn2, 2016-10-07:
Hi, I'm just wondering whats your recommendation , is 2.0 Platform pretty reliable, I am building a machine with a mate a wee router. Should we be brave and use Platform or be conservative and use Classic 2.0 ? I know its really up to use to decide but your guidance on this is appreciated. Cheers and thanks for all your great work.

jahnj0584, 2016-10-07:
Use the platform

winder, 2016-10-07:
The differences are mainly cosmetic. The platform literally uses the UGS Classic as a library, so all the machine communication code is identical.

closed 2016-10-07 by winder

splitn2, 2016-10-09:
thanks for the advice guys. I think I have a dud Mega it kind of locks up, now I have tried with with both USG platform and classic  and also bCNC I am getting similar problems. cheers

jahnj0584, 2016-10-09:
Did you send $X?

splitn2, 2016-10-10:
I'll have a closer look tonight..... thanks @jahnj0584