winder/Universal-G-Code-Sender #498: tool change question

Encarnacao, 2016-10-08:
i would like to know how you guys do  tool changes
from what i see to do  tool changes i would put a M00 on the part of the code that i want to change my tool, but i notice that when the  program stops the motors  will stay with out tension so that means that any  litle move and you will move the plate off the current position,is there any way to make it motors be with tension when you do the tool change to avoid any wrong moves?
How you guys do it?

jahnj0584, 2016-10-08:
$1=255 will send constant power

You can also just rehome after the change. I use seperate files per tool.

Encarnacao, 2016-10-08:
so i would do homing after change tool and then just unpause and goes to where it was?
do you use the $1=255?
i knew that you could use the $1=255 but i dont want to be in constant power motors, imagen i have to go somewhere and forgot machine open then motors would be in constant power, wouldnt be a good idea

jahnj0584, 2016-10-08:
The home recalibrates the machine in case you move it.

That's fine.... The motors don't burn out or have an hourly life span to
them. You could add $1=25 to the end of your file if your so worried.

Encarnacao, 2016-10-09:
@jahnj0584   so in tool change you sayed i could re home after tool change and then  hit resume?
i have tryed make a M00 in where i want to stop to make the tool change, so how then i can send home, i have tryed add the on code $H after the M00 so when i hit resume wold send it home, but off course doesnt work is there any other code to do this?

closed 2017-06-21 by vMeph