winder/Universal-G-Code-Sender #500: File duration information

jahnj0584, 2016-10-11:
Would it be possible to add things such as: 

Run time for last time file was sent (as in saving it within the gcode or into the title)

Run time saved after file is sent (i charge for jobs but the timer resets itself sometimes before i can check it - leave time until new file is loaded, or a console message such as ***Finished sending testfile.nc in 10:36***)

Total run time: tally how long the machine was running today/this week/all time. Not necessary but would be a cool feature (some music players do this)

winder, 2016-10-12:
All set, the finished sending message: `**** Finished sending file in 12:34:56 ****`

closed 2016-10-12 by winder

jahnj0584, 2016-10-12:
I JUST opened the version from the last hour. Time to redownload :P thanks a bunch for the super fast turnaround

commit 5ee10130 (2016-10-12): Add send duration to stream complete message. Closes #500