winder/Universal-G-Code-Sender #501: Work 0 resetting after job cancel

jahnj0584, 2016-10-11:
If you stop or soft reset the machine it resets your work offsets to some old value.... I've broken a bit or two from this already! I caught it a few times now, I'd be at Z0 but if i did a soft reset or a stopped a file itd change to some other value.

winder, 2016-10-12:
Do you use G92? If yes you might want to try G54-G59 instead which is saved across machine resets.

jahnj0584, 2016-10-12:
How does that work? Instead of hitting reset or using my probe command I change it to g54?

winder, 2016-10-12:
The best documentation I've found is on the [shapeoko wiki](http://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems). 

Basically after a probe you would do something like `G10 P1 L20 X0 Y0 Z<probe thickness>`.
The P parameter chooses which coordinate G54-G59 to update and that location is saved in the arduino.

jahnj0584, 2016-10-12:
oh that is clever. Ive been slowly adding more fixtures for repeat jobs to instead of going to my hard limits, then jogging over to their written coords I could just keep them as an offset. Any way we could make them easier to use in UGS? Maybe 2 buttons per G5X (like macro, only show used+1 option) which would be Set G54 and Go To G54?

closed 2016-11-17 by winder