winder/Universal-G-Code-Sender #502: Feed override

jahnj0584, 2016-10-12:
Sonny just said 1.1 will probably be released tonight so was wondering how long we'll have to wait for the feed over ride and current feed rate settings to be released as well. I know he sent out 1.0 early to the devs.

chamnit, 2016-10-12:
@jahnj0584 : Grbl v1.1 overrides works with UGS platform. There is a menu item you have to enable to view the controls.

winder, 2016-10-13:
@jahnj0584 They're already in the platform (the prototype has been in since march I think). The interface could probably be tweaked a little, but it's working.

[image]

closed 2016-10-13 by winder

jahnj0584, 2016-10-13:
My panel on windows isn't like that. I have the -- - reset + ++ for each and no numbers alongside