winder/Universal-G-Code-Sender #503: Over ride over applies

jahnj0584, 2016-10-13:
Lets say your max speed is 600, and the program is at 500. If I apply +20%, it'll max out. I can still click through, adding up to +100% (1000 total) but the machine will still move 600. The issue is, when you go to lower the value there is no real world speed effect until you are again at 20% or lower (so 8-80 clicks of no effect and potential damage). Not sure how you can limit this, but it's something that should be done. Cheers

chamnit, 2016-10-13:
@jahnj0584 : This should be a Grbl issue, rather than a UGS. True that 600 would be maxed out at 120%, but other moves may be programmed to be less than 500. For example, if at 400mm/min, then a 150% override would max it out. You also have to consider that a 600 max is not 600, when you move in diagonal directions. This can be up to 1.414 times that if the other axis is also 600.

For most usage cases, feed rate override is only for FEED moves, where you are actually cutting the material. You are generally moving much slower than the max rapids rates. At full 200% feed override, you will likely not reach the rapids rates anyhow.

winder, 2016-10-13:
@jahnj0584 I'm not sure there's anything a GUI can do about this. You can imagine that in the same program you might have some parts where the feed is much lower. For instance a Z plunge could still see effect at 200% even if you're straight cuts already maxed out at 120%.

A better solution would probably be for me to finish updating the status messages to display the current feed rate, so that you could fine tune the speed more accurately without "accidentally" going 8 clicks past your maximum.

You should post this question on the gnea/grbl board, I think @chamnit is looking forward to feedback like this.

jahnj0584, 2016-10-14:
I xposted the issue, so up to you which one of you wants to figure it out.

Maybe I machine funny, but I have a max rapid of 4200mm, and often am cutting at 3700 or so. I'll be using over ride to see if I can actually cut full pace and maintain accuracy and square-ness. UGS has a separate FOR for rapids/spindle/feeds so that also shouldnt be an issue. Im not sure what the solution is really, besides doing some computation of (feed rate * OR%) <= $110/111/112 value

closed 2016-10-14 by winder