winder/Universal-G-Code-Sender #504: Potential GUI Idea

jahnj0584, 2016-10-14:
Just messing around. Puts everything on one screen without tabs, but even bigger buttons then currently so should be easier to use on tablets or low res screens. Lemme know what you think, and if this is a possibility. 

[image]

DOUG888, 2016-10-14:
I like the Idea, one screen with all the info displayed is better than jumping into tabs to find the info, luv the bigger buttons, can’t remember all the times I have clicked stop instead of pause, but don’t forget the Macros, there could be a lot of those dependant on the end user, I currently have 6 but most likely more will come as I learn more myself.

 

 <https://cloud.githubusercontent.com/assets/17396492/19373240/e880dce8-9198-11e6-9417-014c8705b67d.png> ugs ui

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

winder, 2016-10-15:
One of the main drivers towards using a GUI framework like netbeans was to avoid the "one-size-fits-all" design. You can move the windows around however you like. Most of the tabs are designed to be standalone widgets that you can mix and match for this reason. It also makes it much easier to add new widgets, since they don't have to be permanently embedded into the existing UI.

In that spirit I'm curious specifically at the widget level, which ones should change or be combined? Seems like the new overrides UI could be improved (not surprising, I whipped that one up quickly).

jahnj0584, 2016-10-15:
I kind of expected that answer. What works for me definitely won't work so well for someone who cuts differently. I really hate the mach3 UI as it's over crowded and has unnecessary parts.

 I think if anything, you should seperate every module and have it be an option to join them (say, jog controller with machine coords or macros+common actions) so they can be moved together. 

Here is the closest I was able to get (can't upload to github on mobile, and shop pc getting Internet today) 
http://imgur.com/qkjFLnN

The issue is some modules have a "min size" and therefore can't be more compressed. My suggestions for each module are marked already, but I can list them out if you'd like a task list to run through? Im starting to feel like I'm the only one complaining for new features in this forum :P (Don't worry, i will be donating soon enough)

DOUG888, 2016-10-15:
Hi Guys..
After reading your posts I have found how to move , resize, float etc the individual windows....but, when i dock them they do not stay in the same place or size that i had set, what's the trick???   ;-)
Doug

jahnj0584, 2016-10-15:
You drag the title bar, without popping them out. Like where it says
console, drag the actual word tab and the orange square shows where itll be

DOUG888, 2016-10-15:
Will, your comment  
> In that spirit I'm curious specifically at the widget level, which ones should change or be combined? Seems like the new overrides UI could be improved (not surprising, I whipped that one up quickly).
  
In JOG, I think if you moved the XYZ buttons to the RH of the STEP Size box you would have the same minimum width as the OVERRIDE UI, then you could blend the two together vertically, then they could be compressed vertically to MIN height, this would create more space in that column for other UI, etc, etc.

DOUG888, 2016-10-16:
OK, got that to work, thanks.

Have run out of vertical room in my LH column, even with all the UI pushed up to minimum, I hope Will modifies the JOG UI as suggested as it will give more vertical available space.

 

https://www.avast.com/antivirus

jahnj0584, 2016-10-16:
> my6S9hoSpbA8ks5q0WRfgaJpZM4KWkLM> . <https://github.com/
> notifications/beacon/AT7z7tU3vGKBhjs_0mJCBgmMLwL4-

You can also just use keyboard commands to jog, such as ctrl+arrow keys and
place it wherever. I have no issue with space

DOUG888, 2016-10-16:
Yep, got the movement working, as you described but still out of space LH column,, I think I am trying to get too much in and am exceeding the available vertical space.

I have used the “SIZE GROUP” on each of the UI, if I understand this command correctly, it reduces the UI to its designated Minimum, in this case vertically, horizontal would be set by the “Speed Overrides” UI which is the widest. Screen resolution is at Max. if that makes a difference.

 

https://www.avast.com/antivirus

DOUG888, 2016-10-16:
Preamble: I did a lot of programming in Visual Basic for many companies i worked for and became proficient in Maximizing available space without cluttering thinks up, so these are graphical, space optimizing suggestions, not criticisms, please take them as such.

Ok..have been experimenting with relocating the UI, here are some suggestions that may be useful:-

firstly the "Speed Overrides" UI appears to be the longest with dimensions of 309 x 112. With this in mind, take 309 as the Maximum UI design width for "Common Actions", "Jog Control", Send Progress", and DRO"
This does not mean that they can't be stretched further, just keep the buttons inside 309.

Now, In "Jog Controller", put Step Size, Inches and Millimeters on one line, if its tight, the step size box only needs 4 characters, this would still give 9999MM of step. Move the JOG buttons up. new area now approx 309 x 190.

Next. In "Send Progress", move "Rows", "Sent" and Remaining" to the LH side, abbreviate text to read "Rows", "Sent" and Remaining", now move "Est Time Remaining" and "Duration" up to the top RH side, Change "Estimated"to "Est." new UI size approx 309 x 75.

One other thing that could be done is to move the "Help" button to the RH side of the "Common Actions", new size for UI approx 309 x 160.

Results are that all the above mentioned UI now have the ability to combine into one column if required and will Graphically align as well.

As I said, these changes are not a criticism, luv the work so far, just my way of helping with what i can. 
 I have attached a pic, unmodified UI, but you get the idea.
Please note that I seem to be out of space at the bottom of the screen, can't seem to get this corrected, my guess is that i am trying to get too much into one column, these suggestions will eliminate that and give space to spare.
[image]
Doug
Question: In Workflow Helper, what is Toolname and is it necessary, more space for Filename. I realize that i can adjust column widths.

jahnj0584, 2016-10-16:
I also think the coord widget should be more like GRBLPanel/mach,with
machine being a smaller number. It's only used for setting offsets or jigs.

Step size could go in the center of the Y/X jog buttons! And the units
could be a selection under the Z buttons or hidden in settings. I don't
like mixing up my units and only run metric.

DOUG888, 2016-10-16:
Yes, Unit selection in Controller Options makes good sense, typically it’s a set and forget,  Step Size tucked in the middle of XY jog could work, worth a try.

 

https://www.avast.com/antivirus

DOUG888, 2016-10-16:
Will
Just loaded latest UGS platform...the width of the "Speed Override" has been reduced and reconfigured, but the other suggestions from Jahn and myself should still be Viable.

breiler, 2020-11-01:
There is a lot of ideas in this issue that either has been fixed or needs to be seperated to their own issues. This is a bit too big to work with. 

If there is still stuff you want to have fixed create new issues.

closed 2020-11-01 by breiler