winder/Universal-G-Code-Sender #507: Translation to Russian problem

Werwolfx, 2016-10-15:
When I switched on the Russian language, all menu items are gone and can not be switched back to English
Where are the Setup? I use Linux Mint x64

Werwolfx, 2016-10-15:
I found MessagesBundle_ru_RU.properties contains many blank lines, I now understand why the menu is not displayed after the switch. I know Russian language, and I can help you to translate into Russian. You need this help?

jahnj0584, 2016-10-15:
Yes he does. Will can link you to the webpage when he's on next

winder, 2016-10-15:
@Werwolfx Localization is managed here: https://poeditor.com/join/project/2J2hB5I41Z

Someone started working on it fairly recently but never finished. Thanks for the offer!

closed 2016-10-28 by winder

SadovovAlex, 2016-11-10:
HI

i translated to Russian lang 100%

https://poeditor.com/projects/po_edit?id_language=134&id=52743

Please see it

winder, 2016-11-10:
@SadovovAlex Thanks! There were a couple terms I added recently and forgot to import to poeditor. Could you update those last terms? I'll get them into the project later this morning

SadovovAlex, 2016-11-19:
@winder Sorry for the long absence...

I translated the remaining rows to russian language.