winder/Universal-G-Code-Sender #508: Grbl 1.1 name additions

jahnj0584, 2016-10-15:
The $$ results are nameless now. Can you add back the identification tabs for each value, or maybe a chart style tab/window you can add as a plug in to edit and see all the values? I think grbl panel has that system in place

winder, 2016-10-15:
All set, this was from a GRBL change to save program space.
I'm using the [data file here](https://github.com/gnea/grbl/tree/edge/doc/csv) which actually has longer descriptions than the original:
```
$0=10 (microseconds, Step pulse time: Sets time length per step. Minimum 3usec.)
$1=25 (milliseconds, Step idle delay: Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
$2=0 (mask, Step pulse invert: Inverts the step signal. Set axis bit to invert (00000ZYX).)
$3=0 (mask, Step direction invert: Inverts the direction signal. Set axis bit to invert (00000ZYX).)
$4=0 (boolean, Invert step enable pin: Inverts the stepper driver enable pin signal.)
$5=0 (boolean, Invert limit pins: Inverts the all of the limit input pins.)
$6=0 (boolean, Invert probe pin: Inverts the probe input pin signal.)
$10=255 (mask, Status report options: Alters data included in status reports.)
$11=0.010 (millimeters, Junction deviation: Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
$12=0.002 (millimeters, Arc tolerance: Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
$13=0 (boolean, Report in inches: Enables inch units when returning any position and rate value that is not a settings value.)
$20=0 (boolean, Soft limits enable: Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
$21=0 (boolean, Hard limits enable: Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
$22=0 (boolean, Homing cycle enable: Enables homing cycle. Requires limit switches on all axes.)
$23=0 (mask, Homing direction invert: Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
$24=25.000 (mm/min, Homing locate feed rate: Feed rate to slowly engage limit switch to determine its location accurately.)
$25=500.000 (mm/min, Homing search seek rate: Seek rate to quickly find the limit switch before the slower locating phase.)
$26=250 (milliseconds, Homing switch debounce delay: Sets a short delay between phases of homing cycle to let a switch debounce.)
$27=1.000 (millimeters, Homing switch pull-off distance: Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
$30=1000. (RPM, Maximum spindle speed: Maximum spindle speed. Sets PWM to 100% duty cycle.)
$31=0. (RPM, Minimum spindle speed: Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
$32=0 (boolean, Laser-mode enable: Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
$100=250.000 (step/mm, X-axis travel resolution: X-axis travel resolution in steps per millimeter.)
$101=250.000 (step/mm, Y-axis travel resolution: Y-axis travel resolution in steps per millimeter.)
$102=250.000 (step/mm, Z-axis travel resolution: Z-axis travel resolution in steps per millimeter.)
$110=500.000 (mm/min, X-axis maximum rate: X-axis maximum rate. Used as G0 rapid rate.)
$111=500.000 (mm/min, Y-axis maximum rate: Y-axis maximum rate. Used as G0 rapid rate.)
$112=500.000 (mm/min, Z-axis maximum rate: Z-axis maximum rate. Used as G0 rapid rate.)
$120=10.000 (mm/sec^2, X-axis acceleration: X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
$121=10.000 (mm/sec^2, Y-axis acceleration: Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
$122=10.000 (mm/sec^2, Z-axis acceleration: Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
$130=200.000 (millimeters, X-axis maximum travel: Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
$131=200.000 (millimeters, Y-axis maximum travel: Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
$132=200.000 (millimeters, Z-axis maximum travel: Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
```

winder, 2016-10-15:
As for a settings editor... there actually is one in the classic GUI. I've been meaning to bring it to the platform for ages.

closed 2016-10-15 by winder

winder, 2016-10-15:
I still need to do this for error codes and alarm codes, which no longer return a description.

Encarnacao, 2016-10-15:
Awsome job
@winder off the topic  is it possible that when files off gcode is sending would tell the N line numbers
not sure if would be possible and if would be a good idea?

jahnj0584, 2016-10-15:
Hmmm. That data file looks awefully busy when you're just trying to look for one setting to change. Maybe make that in the help menu, and use the old "names" from 0.9 like (X axis - steps/mm) and nothing more. 

Please do, and let me know when to test it out :). I'm 100% set on the platform from here on out. 

I think he means to show what the last line sent was? This basically falls under rows sent/remaining if it was a little bit closer to real time

commit 9cb889af (2016-10-15): Utilize grbl 'setting_codes' file for setting descriptions. Closes #508