winder/Universal-G-Code-Sender #509: Visualize table view

Encarnacao, 2016-10-15:
For exemple my settings for my machine are 432 by 338 in size of my table max travel
on the visualize i see that max travel is 11 by 11 on design table squares, is it possible that the table side would be updated in size of the specifications that we set in grbl settings.
$130=432.000 (x max travel, mm)
$131=338.000 (y max travel, mm)
or that we could specify the table size?
for when we want to move machine to a specific point we could just righ button on area of the table  and would travel to there

jahnj0584, 2016-10-15:
Are you using homing switches?

Encarnacao, 2016-10-16:
yes

vMeph, 2016-11-10:
@winder any toughts about make this happening, maybe set on the options/controller options like you did for feed speed scaler edit, maybe do a table view space edit, and we could set the table size acording to machine size that would change the size of our table view, not sure how would then be able to see the  squares table divide,not sure how other fells about this option, but in my opinion could be good, if i want to move machine to a specific point of my machine table could be easier just by see on  visualizer table view and  right button click and send machine to the area you want other then jog or send a coordinate

winder, 2016-11-10:
@vMeph this isn't very high on my priorities, I left it open as a suggestion because I like the idea. I want to finish up 2.0 for an initial release (pretty close now), then work on some more advanced features like auto leveling and tool change scripts.

vMeph, 2016-11-10:
Awsome , cant wait for auto leveling feature to come out hehe 👍

breiler, 2018-06-15:
I've added a new renderable that will be active if soft limits are enabled. It will show the machine boundaries using the max travel settings in GRBL.

There is a short demo here:
https://youtu.be/MshHEGVBxzQ

closed 2018-06-15 by breiler