winder/Universal-G-Code-Sender #511: visualizer problem

ufraa, 2016-10-21:
Hi all, 
i have a problem with visualizer. in the config menu all items in the list are disappeared and set to the same color, i can change them but when i press apply it resets to the previous color. if i change the last item color and press apply it sets the color for all the items
[image]
[image]

Encarnacao, 2016-10-21:
no clue what could be did u try maybe delete temp files and reinstall to see if fixes?

winder, 2016-10-22:
Very strange. My guess is that somehow the configuration files were corrupted and things are crashing when they try to load/use them.

In the about menu it tells you the location of the **Cache directory** and **User directory**. Try deleting those directories.

winder, 2016-10-28:
@ufraa were you able to try removing the cache and user directories?

ufraa, 2016-10-29:
Yes. The problem was one of the file and the italian localization
Thank you so much

winder, 2016-11-10:
@ufraa this seems to have something to do with localization, I'm not quite sure why though.

For now if you would like to contribute Italian translations, that should prevent the issue from happening for now. You can join the localization team here: https://poeditor.com/join/project/2J2hB5I41Z

closed 2017-05-06 by winder