winder/Universal-G-Code-Sender #512: Having issues with GRBL1.1 and sender with laser

maudette01432, 2016-10-21:
I'm having an issue with UGS latest nightly build when sending to GRBL 1.1d on a laser it seems to send the moves and spindle speed separately with the spindle speed as the second command.  What this results in I belive is that the laser is not turned on/off at the proper times.  I have pieces that get cut that shouldn't and pieces that don't get cut that should.  Same GCODE works with GRBL 0.9 and UGS 1.0.9.  The moves seem queued up before the laser is turned off or on...it goes through the moves then issues the spindle speed update.  The intent of the GCODE is to move through the segment with the spindle speed and feed specified.  If the command is broken in this way It will go through the move with the current spindle speed and then adjust it.  I could be wrong..but there is no way to disable this "split" behavior in the GUI (it's grayed out).

Config Pane:

[image]

GCODE Example:
[DP_TEST.zip](https://github.com/winder/Universal-G-Code-Sender/files/545124/DP_TEST.zip)

LASER RESULT (note camera rotated 90 degrees - cut is in proper orientation) :
[image]

UGS GUI Display of path:
[image]

What gets sent/returned:
```
ok
ok
Skipping command #2
Skipping command #3
>>> G90
>>> G17
>>> G21
>>> G91Z0
>>> G90
Skipping command #9
Skipping command #10
>>> T1
>>> M3
>>> G1X5.423Y21.359F1000
>>> S0
>>> G3X5.604Y20.594I1.402J-0.071
>>> S1000
ok
>>> G3X6.687Y19.143I6.653J3.835
>>> S1000
ok
ok
ok
ok
ok
ok
>>> G3X7.764Y18.249I3.134J2.681
ok
>>> S1000
ok
ok
>>> G3X9.126Y18.268I0.665J1.22
>>> S1000
>>> G1X9.326Y18.385
ok
>>> S1000
ok
...
```

maudette01432, 2016-10-21:
I have confirmed that this is in fact caused by the splitting of commands in the newest latest branch of UGS.  Same GCODE...Same Firmware...Same Machine.

"NEW" is the cut from sending with the latest nightly build of the GCODE attached above to a GRBL board running Grbl J-Tech Photonics 0.9g J Tech

Old was the same gocde file sent to the same board using the old Java based Universal Gcode Sender 1.0.9.

[image]

This presents a danger to the user as well.  It would appear this was done to enhance error checking/reporting - but unfortunately it's causing problems (and could put someone in danger - laser firing when it should not - spindle not up to speed during a move or vice versa!).

winder, 2016-10-21:
Thanks for the report, nice detective work figuring out the root cause! The command splitter has been modified to avoid splitting the S code into its own line. Could you give the nightly build another try and let me know if that fixes the problem?

Glad you like the new UI. :)

maudette01432, 2016-10-21:
Wow that was fast...I'll give it a poke right now!

maudette01432, 2016-10-21:
And - EPIC...Confirmed it is working.  Now to load GRBL 1.1d back on and finish what I started!

closed 2016-10-21 by maudette01432