winder/Universal-G-Code-Sender #513: UGS Platform tabs organization

Encarnacao, 2016-10-21:
would it be possible to add extra tabs to this area sense theres alot empty space there for a better organization, i wanted to  put there the browser files for exemple for better visible organization but it dont allows to do it

[image]

winder, 2016-10-21:
Yes, the content of the top bar is totally customizable. Right now the contents are in two widgets but many more could be added. Also just like the panels these can be enabled/disabled and moved around.

Encarnacao, 2016-10-21:
so for what i understand they are in seperate widgets so how can i move the  " file browser " to the top?
it dont let me , and i dont see where is enable or disable option

winder, 2016-10-22:
When I said many more could be added I meant that I could create more of them. Right now there are just two.

jahnj0584, 2016-10-25:
yea, that whole tab/bar is "locked" for me.

jahnj0584, 2016-10-25:
The send buttons also should be seperate and larger. I like the idea of
adding the open bar there

winder, 2016-11-06:
Added the file browser:
[image]

closed 2016-11-06 by winder

vMeph, 2016-11-06:
Not home ill test later, thanks alot Will😁

vMeph, 2016-11-06:
@winder  #678 shows has failed