winder/Universal-G-Code-Sender #517: Workflow Event Box opens on Jog

DOUG888, 2016-10-24:
Now running GRBL 1.1
When i move around with Jog I get the workflow event "finished sending last file" popup appearing at each jog, this a annoying ,to say the least :-)
particularly if you are doing small repeated jogs.

This same box also appears when any of the "Common Action" buttons are pressed and the "Macro" buttons.

can this be removed please....

also I noticed that the Z step does not respond to the scrolling arrows in the box, even when typed in, feed rate and xy step are ok.

DOUG888, 2016-10-28:
Now running GRBL 1.1d and UGS platform #671....same issues as above.

winder, 2016-10-28:
@DOUG888 this is on my radar. My test arduino blew up on monday and I'm still waiting on the replacement. You can see an explanation I posted on #522

DOUG888, 2016-10-29:
Ouch!!!  All good Will..new you would get to it ...hope the replacement gets to you soon.
Also..while the test bed is down...I had 2 other issues up, one you asked me to put s a reminder to you..that was on the COM port auto reconnect..I think that is OK, you might want to delete it.

commit 12e717cc (2016-10-30): Fix workflow popup when jogging. Closes #517
closed 2016-10-30 by winder