winder/Universal-G-Code-Sender #519: Not issue but idea ... hmmm

victorhantelmann, 2016-10-26:
Ideea starts with using some Arduino Nano that have GRBL firmware and TX/RX pins 0 & 1 tied to use RS232 to one unique Nano that by using SoftwareSerial Library can manage up to 3 connections.

No mod is needed for GRBL software because 0 & 1 are hardwired to USB (USB is not going to be used here) the default serial is going to be used to some MAX 232 equivalents.

To handle synchronous movement one can use the following in the master Nano firmware (I am working on it):
G0 X10F1000 (Send any valid command)
G4 P0.00001 (Needed for synchronization of all child boards)
$!          (G Code HOLD  CYCLE)
(Did all boards confirmed reception ?)
(WHEN *** YES ***)
(SEND)
$~          (G Code START CYCLE)
(There may be synchronization problems here because each $~ is sent to each board and is dependent on Serial Communication Speed (may be use a common pin) <decision after testing>)

ONE WAY: The only mod that GRBL needs, is to differentiate between "$~" that erroneously creates a listing of Serial Help from GRBL firmware and regular GRBL "$" that should create the Serial Help listing.

SECOND WAY: While "~" in Arduino Serial Monitor responds "ok" and "!" followed by "~" responds "ok ok" and "!" holds, while waiting for "$~" or "~". UGS is forcing the user to press "Pause" double on the GUI...

Those described above (even if SECOND is a good safety feature), stops me from finding a logic algorithm to program on the master without modifying either "$~" to be different then "$" in GRBL either asking you to suspend the double Pause feature that follows "!" and "$!".

What good would come from this? one might ask ... Well a freeware firmware that will overcome the 3 axis limitation of GRBL I hope ...

Also I want to thank you all for the great, great, Great, GREAT work !!!!!!!! that you did in developing UGS in both flavors. It really is a beauty !!!

winder, 2016-10-26:
Sounds like a neat project. Hopefully we'll get more than 3 axis control out of GRBL once Sonny starts the successor to the GRBL-Arduino.

For now, UGS could be extended to support custom commands for a controller like this. You can see some of the extension points described here:
http://winder.github.io/ugs_website/dev/backend_development/

victorhantelmann, 2016-10-27:
Thank you for the info, I will start learning. Communicator and Listener look promissing... I intend to keep GRBL on Nano because for anyone this is the cheapest most affordable arduino. I worked on many other uC PCB's much more capabile

victorhantelmann, 2016-10-27:
capable* and expensive, but "keep it simple" concept :)

victorhantelmann, 2016-10-27:
Sorry, I was on my smartphone when I made the previous comments and I got a call that interfered with my writing ...

Anyway, I have only one more question about the place of this project.

I intend to make it all available on git as soon as I have the foundation for the master's serial tested.

What is advisable? I use and please consider following mapping:
UGS                         GRBL
MASTER - X Y Z -> CHILD 1 X Y Z (GRBL unchanged, default settings)
MASTER - A B C -> CHILD 2 X Y Z (GRBL unchanged, default settings)
MASTER - U V W -> CHILD 3 X Y Z (GRBL unchanged, default settings)
So UGS will see XYZ ABC UVW and 3xGRBL will see only XYZ

I consider this mapping to be more related to UGS project then to GRBL even if in fact it is a firmware for Arduino MASTER side (allowing only 3 childs for the moment because of SoftwareSerial Library limitations).

Why I am considering this? Because UGS-Platform has the new graphical that can do so much more ... (even if all depends on user setup for his machine, at least the following I will keep as true: XYZ linear vectors, ABC rotational vectors, UVW debate on cinematic <tool change, pick and place pneumo -mechanic head, etc.> so UVW must NOT be for the moment represented in the graphic view, only DRO's).

May be I am going too far with this, as it is a hobby open loop system...

I am thinking that this MASTER firmware should in the future be included in your project as an optional feature.

That is why I will study all the info you kindly gave me and I will get back to you with the initial firmware for the master.

Thank you again, wish you the best !!!

closed 2016-11-17 by winder