winder/Universal-G-Code-Sender #520: Auto leveling on UGS platform

Radiu, 2016-10-26:
for a long time i hear that people would love to have UGS with auto level option
this could make UGS more atractive and bring more people into comunity, alot people does pcbs using eagle cad and ends up moving into other programs that alows you to have other options like(chilipepper,bCNC)
Winder has been doing a great job on UGS platform it have grown alot in terms of visual etc.. however it needs to add some other cool features and auto leveling is one of them

any ideas when would this ever be possible?

Encarnacao, 2016-10-26:
no idea when this would be implmented, but for sure that would give UGS a new edge
chilipeper idea is a great thing, specialy with the eagle brb files import
i myself also looking foward for this implemetation to happen in UGS platform

winder, 2016-10-26:
Auto leveling is something I've been working towards for a long time. In addition to the platform being a way to add features like this to the GUI, there are a lot of changes under the hood to make processing gcode better and more configurable. Once UGS platform is cleaned up for a proper release I want to work on things like this.

Someday I'll get to auto leveling. #267

jahnj0584, 2016-10-27:
If chillipeppr is open source can Will not just use the source code and add
it as a widget to UGS? A lot of the wood and metal guys have no need to
auto level

Encarnacao, 2016-10-27:
if that would be possible would be a great idea !!!!
@winder any toughts how to make it happen?

winder, 2016-10-28:
The actual logic to do this is fairly simple, it's a matter of generating a height map and shifting each Z movement to align with that height map (there are some nuances to this, but UGS already has the ability to split things like arcs into line segments). So the only challenge left is integrating with the system, which is completely different in these different senders. A lot of the work I've done in the past year are going to make that integration much simpler than it would have been.

closed 2016-10-28 by winder

luisluna2, 2018-07-29:
YES, NOT WORK NOW.

BEST TRY WITH GRBLCONTROL:

https://github.com/Andypos/grblControl
https://www.youtube.com/watch?v=tK5Amg-Ye58